less-smog/hardware #3: Optional 5V regulator is too hot at 12V

NiceCircuits, 2018-04-27:
When optional 5V regulator is used and 12V power supply is connected, 7805 temperature reaches >90*C.
Proposed solution: use 5V DC/DC converter module, like this one:
https://www.ebay.com/itm/5W-9V-12V-24V-To-5V-DC-DC-Step-Down-Converter-Module-Replace-TO-220-7805-LM2596/222641567186

jasiek, 2018-04-27:
This is a drop-in replacement, right?

NiceCircuits, 2018-05-01:
Yes, but I would like to check if it will fit on the PCB without problems and remove 7805 from documentation to discourage anyone from using it ;)

NiceCircuits, 2018-05-01:
Another options:
https://www.ebay.com/itm/DC-DC-7V-28V-9V-12V-To-DC-5V-3A-DC-DC-Step-down-Buck-Conveter-Module-Chip-Tools/222913561934
https://www.ebay.com/itm/3A-5v-9V-12v-To-3-3v-5v-DC-DC-Buck-Step-down-Power-Supply-Converter-Module/173273469891
but only versions without regulation by potentiometer. With regulated, somebody will almost certainly set invalid voltage and cook the board ;)

jasiek, 2018-05-02:
This is an optional component anyway - let's just make sure that the DD4012SB_5V fits on the board and be done with it.

closed 2018-05-22 by jasiek