less-smog/hardware #4: Replace current SMD LEDs with THT variants

jasiek, 2018-05-02:
Given we'd like these to be assembled by students, etc - I think it makes sense to pre-assemble the SMD components and leave a few ones which are easy to solder by hand as an exercise. I'd like us to replace the LEDs with THT variants - those should be cheaper anyway.

Let's try and pick something which is widely available.

NiceCircuits, 2018-05-20:
We need common anode version. Wider angle will also be better (60deg or more). Proposed:
https://www.tme.eu/en/details/ostapa57e1a/tht-leds-5mm/optosupply/ £0.185/pcs @100pcs.

We can also try some cheap AliExpress LEDs, but they are all 25-30deg - maybe that will be enough?
https://www.aliexpress.com/item/Hot-100-pcs-5mm-4-pin-RGB-LED-Common-Anode-Red-Green-Blue/32830737328.html

jasiek, 2018-05-20:
The LEDs I got from China are 25 degrees, and they seem fine. Keep in mind
that the transparent part of the case will diffuse the light quite a lot.

I'm more concerned about the Red channel operating at a different voltage
compared to green and blue. Are these consistent enough throughout 5mm LEDs
that we can just assume this is a constant across all potential LEDs?

NiceCircuits, 2018-05-20:
Yes, it's common that red LEDs have ~2V voltage drop and bright green and blue ~3V. I don't expect other case.

closed 2018-05-22 by jasiek