less-smog/hardware #6: Is it possible to hook up an SX1276 LoRa module as an add-on given how many pins we have?

jasiek, 2018-05-19:
https://www.aliexpress.com/item/2pcs-868MHz-super-low-power-RF-LoRa-module-SX1276-chip-Long-Distance-communication-Receiver-and-Transmitter/32859550198.html - let's see if we have enough pins left to be able to add this board as an add-on.

NiceCircuits, 2018-05-21:
After pinout consolidation, we have two I/O left. More will be available if programming/debug UART is merged with PMS UART, but it will make debug harder - will not be implemented in this version.
For testing, we can connect another kind of LoRa module with UART communication. I will break out the two free pins to headers.
https://pl.aliexpress.com/item/Free-shipping-2pics-CDSENET-E45-TTL-100-20dBm-3km-SX1276-868MHz-LoRa-RF-Transceiver-Module/32791688505.html

jasiek, 2018-05-21:
👍

closed 2018-05-22 by jasiek